smartmicro/smartmicro_ros2_radars #2: arm64 Support

JWhitleyWork, 2022-01-21:
The release of Smart Access provided with the driver is built for `amd64`. Is it possible to add the libraries for `arm64` support to the driver and have CMake link to the appropriate ones based on detected architecture?

smartSRA, 2022-01-25:
> The release of Smart Access provided with the driver is built for `amd64`. Is it possible to add the libraries for `arm64` support to the driver and have CMake link to the appropriate ones based on detected architecture?

The current Smart Access doesn't provide support for `arm64`, however in the newer versions we do offer such support. It is planned to adopt the driver to include the new Smart Access version.

JWhitleyWork, 2022-01-26:
@smartSRA Is there an expected release date for this new version? Will it work with this existing ROS2 driver or will modifications need to be made? My client needs this as soon as possible for a technology demonstration.

smartSRA, 2022-01-26:
@JWhitleyWork There isn't a release date yet. With the newer smart access, modifications would need to be made to the driver.
As this would be a major update than.

However, I will try and see if we can provide arm64 support for the current version that the driver is using and get back to you.

JWhitleyWork, 2022-01-26:
@smartSRA Thank you!

smartSRA, 2022-01-28:
@JWhitleyWork Unfortunately this would not be possible for the current version.

JWhitleyWork, 2022-01-29:
That is unfortunate. Is there any particular reason why this couldn't be built for `arm64` hardware?

smartSRA, 2022-01-31:
We have a new pipeline for the newer version and all platforms supports are to be the features of the upcoming version. The pipeline for smart access v3xx has been discontinued and it was not intended to be run on `arm64` as there was no such requirements. 

I have already taken your request into account and this is also the first request for an `arm64` support so we have it now in our feature list. As soon as we get the arm64 support for smart access, I will update/release the newer version of the driver which would than have the arm64 support. However, this is going to take some time.

JWhitleyWork, 2022-01-31:
Thank you for the information. My client has `arm64` hardware and three UMRR-96 Type 153 radars so let me know if we can be of any help with testing.

AllanGErnst, 2022-02-13:
> We have a new pipeline for the newer version and all platforms supports are to be the features of the upcoming version. The pipeline for smart access v3xx has been discontinued and it was not [...]

We also have UMRR-96 Type 153 radars and arm64 hardware, and are quite desperate to get the umrr_ros2_driver up and running in ROS2 Foxy. Please let us know if there's anything we can do to help and accelerate this.

smartSRA, 2022-02-14:
I have already started updating the driver to include the new smart access v4 and I will create an issue to track this.

smartSRA, 2022-05-04:
The new update of the driver provides the smart access arm libs. Readme mentions how to use them. It is possible to build and run the driver on an arm machine. I have tested it out on a raspberry pi 4. 

Cross building not offered yet and would be included in the next minor release.

rsanchezm98, 2022-05-23:
@smartSRA I have tried the drivers on jetson agx xavier with ubuntu 18 [changing ros2 foxy to ros2 dashing staff] and it can compile, initialize the node, but callback of the data is never called. I checked in wireshark and data is being received perfectly. 

I tried to compile with armv8-gcc_9 but an error was thrown: 
lib-linux-armv8-gcc_9/libumrr11_t132_automotive1.1.1_user_interface.so: undefined reference to 'pow@GLIBC_2.29' [i tried adding -lm on the cmakelists on target_link_libraries but no succeed]

I checked my gcc version 9.4.0 so it should have been able to compile succesfully]. I compiled succesfully with armv8-gcc_7. But then I do not get data as I mentioned before. I do not know if it is yet compatible for jetson agx xavier...

smartSRA, 2022-05-23:
@rsanchezm98 As you mentioned that you have complied successfully using the armv8-gcc_7 libs, it means there are no compatibility issues. 

I would assume that the driver has not been correctly configured.
Could you share your configs and also the syslogs?

rsanchezm98, 2022-05-24:
There are no errors produced and syslog emitted when launching the node. I checked the subscription and publisher creation and they were correctly done. It goes waiting for receiving data [I put a comment and it never gets to the callback]. I see data is being sent to my jetson as I see on wireshark but the node does not do anything. I am able to ping the radar as well. 

This is my config file:
```yaml
---
/**:
  ros__parameters:
    # The client_id to be set in smart_access_config.json, must be a unique integer.
      master_client_id: 1
      # This sets the dev_id in the hw_inventory.json
      hw_dev_id: 2
      # This sets the port in the hw_inventory.json
      hw_port: 55555
      # This sets the iface_name in the hw_inventory.json
      hw_iface_name: "eth0"
      # An array of sensors to subscribe to.
      sensors:
          # As many as 10 sensors all named as "sensor_<number>" in increasing order of numbers,
          # e.g., sensor_0, sensor_1, etc. The list must start with sensor_0.
          sensor_0:
              # The model of the connected sensor.
              model: "umrr11"
              # The client_id of the sensor/source, must be a unique integer.
              id: 100
              # The ip address of the sensor or of the source acting as a sensor.
              ip: "192.168.11.25"
              # The port to be used.
              port: 55555
              # The frame_id to be set to the published messages.
              frame_id: "umrr"
              # Specify the history size.
              history_size: 1
```

rsanchezm98, 2022-05-25:
@smartSRA I am using ubuntu 18 with dashing because there is no stable version for ubuntu 20 in agx xavier. There is a developer beta that we did not test yet. 
The syslogs is not empty, I mean there is no error or warning. I dit not mean the syslog was empty but the ros logs [as the node does not say anything]. I attach you what I receive in the syslog when running the node:

```
May 25 09:22:26 ubuntu smartmicro_radar_node_exe:  Allocate Batch Base major version Waiting to receive result | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/InstructionService.cpp(457)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to find base interface | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/InstructionService.cpp(484)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: getClientUserInterfaceData() NOT successfully processed | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/InstructionService.cpp(589)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: user interface for client id [0x64] does not exist | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/InstructionService.cpp(94)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Can not fetch user interface info, client[0x64] does not response | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/DeviceMonitorService.cpp(301)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to get user interface info for client[0x64] | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/DeviceMonitorService.cpp(210)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to find Deserializer for port [66] | /code/project/.conan/data/UserInterfaceApi/3.1.0/smartmicro/stable/build/4bc083c04076383de087a06a35ab44d5fe63ca55/src/BaseSerializer.cpp(133)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to deserialize package from clientId[0x64] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/DataServices.cpp(713)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to find Deserializer for port [66] | /code/project/.conan/data/UserInterfaceApi/3.1.0/smartmicro/stable/build/4bc083c04076383de087a06a35ab44d5fe63ca55/src/BaseSerializer.cpp(133)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to deserialize package from clientId[0x64] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/DataServices.cpp(713)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to find Deserializer for port [66] | /code/project/.conan/data/UserInterfaceApi/3.1.0/smartmicro/stable/build/4bc083c04076383de087a06a35ab44d5fe63ca55/src/BaseSerializer.cpp(133)
May 25 09:22:26 ubuntu smartmicro_radar_node_exe: Failed to deserialize package from clientId[0x64] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/DataServices.cpp(713)
```

The ROS2 node says only:
```
[INFO] [smartmicro_radar_node]: Data stream services have been received!
```

smartSRA, 2022-05-25:
@rsanchezm98 With the first look on the syslog I already see the issue. There is clearly a warning here which is send from smart access layer. You are using an old sensor firmware which is not compatible with the driver. Can you ask our support team to provide you with the latest sensor firmware?

Let me know once you have updated the sensor with the new firmware and tried it out.

rsanchezm98, 2022-05-25:
Many thanks! I will reach them out. Hope to be back soon with good news!

rsanchezm98, 2022-05-31:
@smartSRA I have succesfully updated the firmware of my umrr11 type 132 sensors. I tested the new firmware in mi personal computer and succeded. Then I moved to jetson agx xavier again and did not receive data, but the syslog is now different:

```
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Smart Access API version 2.3.0 | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/CommunicationServices.cpp(41)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  COMLIB version 5.1.7 | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/DataServices.cpp(90)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Constructor InterfaceServices | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/interface_services/src/InterfaceServices.cpp(53)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Create eth interface client id[0x64] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/network_services/src/RoutingTable.cpp(416)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Init udp interface link[1] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/interface_services/src/UdpInterface.cpp(44)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Add route Client[0x64]<->Link[1]  | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/network_services/src/RoutingTable.cpp(83)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Init network serices[OK] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/network_services/src/NetworkServices.cpp(82)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Init data services[OK] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/DataServices.cpp(138)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: dlopen [/home/jetsonagx/ros2_ws/install/umrr_ros2_driver/lib/libbasev1.0.2_user_interface.so]  | /code/project/.conan/data/ComLibCommon/3.1.0/smartmicro/stable/build/402369d30c889e0d5a48d4436116171e43225898/src/Types.cpp(58)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Create user interface basev1.0.2 | /code/project/.conan/data/UserInterfaceBaseV1.0.2/2.4.0/smartmicro/stable/build/5d3d35f4cb31d9162ca674bfdea832d3ff928486/out/generated/comm_services/base_v1_0_2/BaseUserInterface.cpp(41)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: Add user iface[basev1.0.2]  | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/UserInterfaceManager.cpp(113)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: dlopen [/home/jetsonagx/ros2_ws/install/umrr_ros2_driver/lib/libumrr11_t132_automotivev1.1.1_user_interface.so]  | /code/project/.conan/data/ComLibCommon/3.1.0/smartmicro/stable/build/402369d30c889e0d5a48d4436116171e43225898/src/Types.cpp(58)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Create user interface umrr11_t132_automotivev1.1.1 | /root/.conan/data/UserInterfaceUmrr11_t132_automotiveV1.1.1/2.4.0/smartmicro/stable/build/5d3d35f4cb31d9162ca674bfdea832d3ff928486/out/generated/comm_services/umrr11_t132_automotive_v1_1_1/Umrr11_t132_automotiveUserInterface.cpp(41)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: Add user iface[umrr11_t132_automotivev1.1.1]  | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/UserInterfaceManager.cpp(113)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: dlopen [/home/jetsonagx/ros2_ws/install/umrr_ros2_driver/lib/libumrr96_t153_automotivev1.2.1_user_interface.so]  | /code/project/.conan/data/ComLibCommon/3.1.0/smartmicro/stable/build/402369d30c889e0d5a48d4436116171e43225898/src/Types.cpp(58)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Create user interface umrr96_t153_automotivev1.2.1 | /root/.conan/data/UserInterfaceUmrr96_t153_automotiveV1.2.1/2.4.0/smartmicro/stable/build/5d3d35f4cb31d9162ca674bfdea832d3ff928486/out/generated/comm_services/umrr96_t153_automotive_v1_2_1/Umrr96_t153_automotiveUserInterface.cpp(41)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: Add user iface[umrr96_t153_automotivev1.2.1]  | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/UserInterfaceManager.cpp(113)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: GetUserIface ifname[basev1.0.2] | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/UserInterfaceManager.cpp(88)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  User If name[basev1.0.2] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/SerializerFactory.cpp(157)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  User If name[basev1.0.2] | /code/project/.conan/data/ComLib/5.1.7/smartmicro/stable/build/d7a18058755baeacc299cb2c117388b98fd0dcd1/data_services/src/SerializerFactory.cpp(157)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Created new shared lib descriptor[/home/jetsonagx/ros2_ws/install/umrr_ros2_driver/lib/libbasev1.0.2_user_interface.so]  | /code/project/.conan/data/ComLib/5.1.7/smart [... cut ...]
May 31 12:55:04 ubuntu smartmicro_radar_node_exe:  Dev Mon Base major version result ready | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/DeviceMonitorService.cpp(279)
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: No connected client callback found for client[0x64] | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/DeviceMonitorService.cpp(204)
```

I am able to ping the sensor and see data traffic on wireshark as well. Hope you can help me!

smartSRA, 2022-05-31:
@rsanchezm98 This looks much better. The sensor and driver have been all properly setup. I cannot come to a conclusion. Could you kindly attach your wireshark dump?

The only two reasons I can see for not being able to receive data could be:

- The driver configuration is not right (your configs do look fine).
-  The package was build for "ROS2 Foxy" and is not compatible with "ROS2 dashing". 

I will look into the wireshark dumps and see if I get a clue if not it is most probably due to the second reason and the only possibility would be to switch to foxy.

rsanchezm98, 2022-05-31:
@smartSRA it seems to be here and independant from ROS2:
```
May 31 12:55:04 ubuntu smartmicro_radar_node_exe: No connected client callback found for client[0x64] | /code/project/.conan/data/SmartAccessApi/2.3.0/smartmicro/stable/build/e5700a0bc7bcac1f2b4a872a1a554fbfb5d2c53b/src/DeviceMonitorService.cpp(204)
```
Besides I have commented the code when the ros2 callback is created and it is succesfully create at runtime, as well as the publisher.

smartSRA, 2022-05-31:
> 

The smart access takes few seconds for calling the callback and this is a warning. Does this repeat or is it printed out only once? If it is not repeating it means that the callback has been found.

Could you kindly attach your wireshark dump?
Also what do you see when you do a "ros2 topic echo /umrr/targets_0"?

rsanchezm98, 2022-05-31:
> The smart access takes few seconds for calling the callback and this is a warning. Does this repeat or is it printed out only once? If it is not repeating it means that the callback has been found. [...]

I finally solved it. Today I selected a wrong host ip [my bad], and now I changed it and I receive and publish data at its rate. Thanks for everything and for telling me to change the firmware version. Now you know that it can be run on ubuntu18 ros2 dashing as well.

smartSRA, 2022-05-31:
@rsanchezm98 I had a hunch that that was the issue which is why I asked you for the dumps. 

Great to hear and you are welcome!

closed 2022-06-08 by smartSRA