smartmicro/smartmicro_ros2_radars #6: Support for CAN Connection

nsr-taylor, 2022-03-21:
Currently I do not have access to the automotive Ethernet connector for the UMRR-96 Type 153 Radar. Does the driver support a CAN USB connection? I have been running the launch file, with some modifications to the radar.template.yaml file, but the node crashes with 'Initialization failed'. Likewise, I was curious if there is any documentation for the Smart Access C++ API? I'd be interested to learn how to set IP addresses for different sensors using this interface.

Thanks in advance!

smartSRA, 2022-03-22:
The driver in it's current state does not support CAN communication. Following the update #4 this is planned to be added as a feature. 
If you could tell me as to what modifications you have made, I could tell the possible reasons for the initialization failures? On a general note, there is not much modifications that you could possibly do other than the numbering and the naming of the interface name. The IP addresses and client ID of the sensors should be unique!

Yes, there certainly is. You would need to contact our busdev as I cannot provide it here. There are two possibilities for setting the ip addresses of the sensors either using the drive recorder or using a smart access API. I have included option two in the driver #1, this has been put to a halt for some internal issues but you could set the IP address following this PR.

nsr-taylor, 2022-03-22:
Thank you for clarifying. Is there an expected release date for the CAN communication update?

Regarding modifications to the parameter file, I have currently removed the second sensor, sensor_1, from the sensors 'block' of the .yaml file and had been playing around with the interface name given my current lack of Ethernet connector. I had changed "hw_iface_name" and "iface_name" both to CAN0 (after setting up a CAN port on my local machine). Likewise, I had tried to set both of the aforementioned parameters to 'lo' to match the default interface name as listed in smartmicro_radar_node.cpp (line 63). The latter attempt works in a sense, but does not actually interface with the radar so all ROS output topics are either empty or meaningless. The standard template with an 'eth0' connection also fails, but this makes sense given that I am not using an Ethernet connection.

Are there any workarounds to getting the wrapper up and running without the automotive Ethernet connector? My setup is identical to the one shown in Picutre 4 of the Smartmicro Project Documentation (i.e. Getting Started User Guide for Automotive and Industrial Radar Sensors). I believe the ROS1 wrapper for the Smartmicro radar supports CAN communication, but if possible, I would like to avoid migrating this project over to ROS2.

Thanks again for both your help and the fast response!

smartSRA, 2022-03-24:
As for the moment there is no planned date for release. There are certain underlying issues which are preventing from integrating CAN.

Yes, that would not work! The driver is designed for only ethernet. It would not work with CAN.

I believe the best possible solution would be to have an Ethernet connector.

closed 2023-09-22 by smartSRA