smartmicro/smartmicro_ros2_radars #12: docker-compose up

CM134, 2022-06-16:
Hi,

I am following the steps in the 'Development' section of the README and run into the following error when `docker-compose up`:

`
ERROR: The Compose file './docker-compose.yml' is invalid because:
services.sensor_0.depends_on contains an invalid type, it should be an array
services.sensor_1.depends_on contains an invalid type, it should be an array
`

I tried to remove the 'depends_on' key, but ran into more errors afterward. 

There was no sensor connected at the time I ran `docker-compose up`.

Thanks for your help.

smartSRA, 2022-06-20:
Hello @CM134,

This appears to be docker/docker-compose version issue. Check that you have the update docker compose version i.e higher than 1.27.0+ 

https://docs.docker.com/compose/compose-file/#depends_on-1

smartBUILDBOT, 2022-09-23:
Fixed in #18

closed 2022-09-23 by smartBUILDBOT