smartmicro/smartmicro_ros2_radars #14: Driver is not compiling in ROS2 Galactic

Grisly00, 2022-09-08:
Hi!

I am having troubles building the driver. I work with ROS2 Galactic inside a docker container on two different systems. 
I tried on ARM (aarch64) as well as x86_64 architecture. I executed the `./smart_extract.sh` file and for the ARM version I replaced all entries of `lib-linux-x86_64-gcc_9` with `lib-linux-armv8-gcc_9`. 
Both fail when building with `colcon build --packages-select umrr_ros2_driver`.

**ARM:**

```
Starting >>> umrr_ros2_driver
--- stderr: umrr_ros2_driver
/usr/bin/ld: cannot find -lcjson
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/smartmicro_radar_node.dir/build.make:160: libsmartmicro_radar_node.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:205: CMakeFiles/smartmicro_radar_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< umrr_ros2_driver [7.60s, exited with code 2]

Summary: 0 packages finished [8.37s]
  1 package failed: umrr_ros2_driver
  1 package had stderr output: umrr_ros2_driver
```

**x86_54:**

```
Starting >>> umrr_ros2_driver
--- stderr: umrr_ros2_driver
In file included from /home/grisi/MOLISENS/molisens_ws/src/drivers/smartmicro_ros2_radars/umrr_ros2_driver/include/umrr_ros2_driver/smartmicro_radar_node.hpp:28,
                 from /home/grisi/MOLISENS/molisens_ws/src/drivers/smartmicro_ros2_radars/umrr_ros2_driver/test/test_smartmicro_radar_node.cpp:16:
/home/grisi/MOLISENS/molisens_ws/src/drivers/smartmicro_ros2_radars/umrr_ros2_driver/smartmicro/include/Instruction.h: In instantiation of ‘bool com::master::InstructionBase<T>::GetConvertValue(uint64_t&) [with T = unsigned int; uint64_t = long unsigned int]’:
/home/grisi/MOLISENS/molisens_ws/src/drivers/smartmicro_ros2_radars/umrr_ros2_driver/smartmicro/include/Instruction.h:183:16:   required from here
/home/grisi/MOLISENS/molisens_ws/src/drivers/smartmicro_ros2_radars/umrr_ros2_driver/smartmicro/include/Instruction.h:189:49: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  189 |         result = *reinterpret_cast<uint64_t *>(&tmp);
      |                                                 ^~~
---
Finished <<< umrr_ros2_driver [3.22s]

Summary: 1 package finished [3.32s]
  1 package had stderr output: umrr_ros2_driver
```

Thanks in advance!

smartSRA, 2022-09-12:
@Grisly00 
Hey,

Regarding the build on x86_64, as I see from what you have posted there is no failure and the driver has been built! There are just complier warnings (I would look into that, did not have them before).

For the armv8, there seems to be linking error however the driver is not using cjson rather nlohmannjson and there is no dynamic linking required with cjson. I am not sure as to why this occurs nontheless, I will try to reproduce it on an armv8 and update you.

Grisly00, 2022-09-12:
@smartSRA 
Hi,
You are right, I did not read the x86_64 output properly.  
Would be great though, if you could look into the ARM issue as this is our main use case with the radar.
Thanks!

smartSRA, 2022-09-12:
> @smartSRA Hi, You are right, I did not read the x86_64 output properly. Would be great though, if you could look into the ARM issue as this is our main use case with the radar. Thanks!

I am not able to reproduce the error that you are getting. I am using the image ``docker pull arm64v8/ros:galactic`` to test on an armv8. I set my libs accordingly for armv8 and the build is successful for me. 

Could you verify that you have properly set the libs for armv8 and that the libs have been downloaded at the correct folder i.e ``/umrr_ros2_driver/smartmicro``? Could you also verify which release are you using?

The ``cjson`` was used in the ``v1.0.0`` of the driver and is not included in the releases ``v2.0.0`` and ``v2.1.0`` and ``v1.0.0`` does not offer arm support.

Grisly00, 2022-09-23:
The driver is now compiling.
The issue was that I initially tried with v1.0.0 before finding out that it does not yet support arm. It was necessary to remove the build and install folders after the update to v2.1.0 before building again. 
Thanks for the help!

closed 2022-09-23 by Grisly00