smartmicro/smartmicro_ros2_radars #15: Compilation and empty topic problems

NilaySener, 2022-09-13:
Hi, i am facing a few errors while running the package. 

- First, when i try to compile i get this error:
```
Starting >>> umrr_ros2_msgs
Starting >>> SensorSimulator
--- stderr: SensorSimulator
/usr/bin/ld: cannot find -lcom_lib
/usr/bin/ld: cannot find -losal
/usr/bin/ld: cannot find -lumrr11_t132_automotivev1.1.1_user_interface
/usr/bin/ld: cannot find -lumrr96_t153_automotivev1.2.1_user_interface
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/simulator.dir/build.make:84: simulator] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/simulator.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
---
Failed   <<< SensorSimulator [0.06s, exited with code 2]
Aborted  <<< umrr_ros2_msgs [2.40s]                               
                                
Summary: 0 packages finished [2.53s]
  1 package failed: SensorSimulator
  1 package aborted: umrr_ros2_msgs
  1 package had stderr output: SensorSimulator
  1 package not processed
```

   I remove the simulator file and it did not give any error while  building. Is this file neseccary to launch node? 

- Then, I run into another problem when I try to launch node. The topic is not visualized on the Rviz screen. When I listen to the topic from the terminal, no message appears in it. I configure the ` radar.template.yaml` file like this:
```
---
/**:
  ros__parameters:
    # The client_id to be set in smart_access_config.json, must be a unique integer.
      master_client_id: 1
      # This sets the dev_id in the hw_inventory.json
      hw_dev_id: 2
      # This sets the port in the hw_inventory.json
      hw_port: 55555
      # This sets the iface_name in the hw_inventory.json
      hw_iface_name: "enp46s0"
      # An array of sensors to subscribe to.
      sensors:
          # As many as 10 sensors all named as "sensor_<number>" in increasing order of numbers,
          # e.g., sensor_0, sensor_1, etc. The list must start with sensor_0.
          sensor_0:
              # The model of the connected sensor.
              model: "umrr96"
              # The client_id of the sensor/source, must be a unique integer.
              id: 100
              # The ip address of the sensor or of the source acting as a sensor.
              ip: "192.168.11.17"
              # The port to be used.
              port: 55555
              # The frame_id to be set to the published messages.
              frame_id: "umrr"
              # Specify the history size.
              history_size: 10

```
When I launch the node, there is no error in the terminal:
```
ros2 launch umrr_ros2_driver radar_rviz.launch.py 
[INFO] [launch]: All log files can be found below /home/nilay/.ros/log/2022-09-13-16-21-37-764950-nilay-40151
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [smartmicro_radar_node_exe-1]: process started with pid [40153]
[INFO] [rviz2-2]: process started with pid [40155]
[rviz2-2] [INFO] [1663075298.036421287] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1663075298.036529534] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1663075298.052637901] [rviz2]: Stereo is NOT SUPPORTED
[smartmicro_radar_node_exe-1] [INFO] [1663075299.831572927] [smartmicro_radar_node]: Radar services are ready.
```
Do you think there is an error in the parameters in the .yaml file? I would be very grateful if you could help, thank you.

smartSRA, 2022-09-13:
@NilaySener 

Hey,

For the first part, the SensorSimulator is not a ros2 package and hence the output. In order to compile the driver and its resources you need to give the following command:

``colcon build --packages-select umrr_ros2_driver umrr_ros2_msgs``. 

For the second part regarding the parameters, yes, there is an issue that I see. The ip address that you have to specify is that of the sensor and not of the interface of your pc/machine. Assuming you are using the default ip address of the sensor(192.168.11.11) and your interface(192.168.11.17). You need to set the ip address of the sensor i.e ``192.168.11.11`` in the param file.

smartSRA, 2022-09-13:
Please make sure the interface that you have specified in the param file ``enp46s0`` is configured to be on the same subnet as the sensor. For instance in your case should be configured to ``192.168.11.17`` and also make sure that you are able to ping the sensor.

markusschratter, 2022-09-13:
@smartSRA 
Hi, it might be a solution to add a COLCON_IGNORE into the simulator folder. It would make the build process more convenient. Especially, if the Smartmicro driver is integrated into a ROS workspace with other packages.

NilaySener, 2022-09-13:
Yes, i can ping the sensor from terminal and also using wireshark. I verfied the interface name also using ifconfig. You can also see the outputs i mentioned:

```
nilay@nilay:~$ ifconfig
br-a93efd67668a: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.49.1  netmask 255.255.255.0  broadcast 192.168.49.255
        ether 02:42:58:91:1a:59  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        inet6 fe80::42:eaff:fe84:3790  prefixlen 64  scopeid 0x20<link>
        ether 02:42:ea:84:37:90  txqueuelen 0  (Ethernet)
        RX packets 34344  bytes 1829616 (1.8 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43138  bytes 220894429 (220.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp46s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b0:25:aa:4b:c3:01  txqueuelen 1000  (Ethernet)
        RX packets 196016  bytes 47569163 (47.5 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 15402  bytes 2764769 (2.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 772277  bytes 100355364 (100.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 772277  bytes 100355364 (100.3 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlo1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.165  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::eff8:1f6e:b27d:458c  prefixlen 64  scopeid 0x20<link>
        ether 5c:e4:2a:8c:a3:16  txqueuelen 1000  (Ethernet)
        RX packets 1332049  bytes 1624492255 (1.6 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 562026  bytes 83333699 (83.3 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
```
nilay@nilay:~$ ping 192.168.11.11
PING 192.168.11.11 (192.168.11.11) 56(84) bytes of data.
64 bytes from 192.168.11.11: icmp_seq=1 ttl=128 time=0.228 ms
64 bytes from 192.168.11.11: icmp_seq=2 ttl=128 time=0.293 ms
64 bytes from 192.168.11.11: icmp_seq=3 ttl=128 time=0.299 ms
64 bytes from 192.168.11.11: icmp_seq=4 ttl=128 time=3.55 ms
64 bytes from 192.168.11.11: icmp_seq=5 ttl=128 time=0.289 ms
^C
--- 192.168.11.11 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4062ms
rtt min/avg/max/mdev = 0.228/0.932/3.551/1.309 ms

```

smartSRA, 2022-09-13:
> COLCON_IGNORE

Hey @markusschratter ,

I was aware of AMENT_IGNORE but did not know of COLCON_IGNORE. This is great. Thanks for the tip!

smartSRA, 2022-09-13:
@NilaySener 

You have not configured(assigned an ip address) your interface:

[image]

As I have previously mentioned you need to configure your interface. Set it to the default of ``192.168.11.17``.

NilaySener, 2022-09-13:
@smartSRA its because of i cut the power of the radar at that time. 

```
enp46s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.11.17  netmask 255.255.255.0  broadcast 192.168.11.255
        inet6 fe80::d5ac:3bd9:d725:37d1  prefixlen 64  scopeid 0x20<link>
        ether b0:25:aa:4b:c3:01  txqueuelen 1000  (Ethernet)
        RX packets 244264  bytes 59738112 (59.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16854  bytes 3261241 (3.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
Unfortunately the topic is still empty.

NilaySener, 2022-09-13:
@smartSRA I would also like to mention that with my current configurations, I can visualize the data that comes with the Windows interface of the device, DriveRecorder3. Based on this situation, I can say that my hardware connections are OK.

markusschratter, 2022-09-14:
@smartSRA 
Hi,
no problem! COLCON_IGNORE is sometimes useful! ;)

I have the same issues and do not receive data with version v2.1.0. Wireshark shows packages on port 55555.
[image]

We use following configuration for the radars:
[image]

The ROS2 parameters are correct set. I only observed following behaviour for parameeter iface_name, which can not be set in the param.yaml:

```
ros2 param get /smartmicro_radar_node sensors.sensor_0.iface_name 
String value is: lo
```
Is this correct?

smartSRA, 2022-09-14:
> @smartSRA I would also like to mention that with my current configurations, I can visualize the data that comes with the Windows interface of the device, DriveRecorder3. Based on this situation, I can say that my hardware connections are OK.

Hey,
Could you kindly attach your syslog and wireshark dump so I can have a look in detail?

smartSRA, 2022-09-14:
> @smartSRA Hi, no problem! COLCON_IGNORE is sometimes useful! ;)
>
> I have the same issues and do not receive data with version v2.1.0. Wireshark shows packages on port 55555. [image]
>
> We use following [...]

Yes, your param look fine and the traffic in the dump is also fine. Could you also provide me with the syslogs?

Regarding the sensor interface, yes it should not make a difference.

smartSRA, 2022-09-14:
@NilaySener @markusschratter Could you both tell me which sensor firmware are you using?

smartSRA, 2022-09-14:
@NilaySener I see that the sensor is not responding. 

Which sensor firmware do you have for UMRR96 T153?

The aktuell release of the driver requires firmware bundle ``UMRR96-T153: V5.2.4`` for the sensor. Could you verify this?

NilaySener, 2022-09-14:
> 

> @NilaySener I see that the sensor is not responding.
> 
> Which sensor firmware do you have for UMRR96 T153?
> 
> The aktuell release of the driver requires firmware bundle `UMRR96-T153: V5.2.4` for the sensor. Could you verify this?

OK, where can i check the firmware version of my sensor?

smartSRA, 2022-09-14:
It should be provided to you along with the other documents from our support team. If not you can get it using the DriveRecorder tool by sending the sensor a get status request (major,minor,patch) or you could contact our support an ask them to provide you with the firmware bundle ``UMRR96-T153: V5.2.4`` and if using ``UMRR11-T132: V5.1.4`` and flash your sensor using the DriveRecorder with the new firmware.

NilaySener, 2022-09-14:
> @NilaySener I see that the sensor is not responding.
> 
> Which sensor firmware do you have for UMRR96 T153?
> 
> The aktuell release of the driver requires firmware bundle `UMRR96-T153: V5.2.4` for the sensor. Could you verify this?

I checked the version and it was V4.1.1, I could update the V5.2.2 and still got the same results. I guess V.5.2.2 is not compatible either. Now, I will try to change it when I receive the V5.2.4 firmware. I will update here soon.

jeremyroy, 2022-09-19:
I had this same problem - I couldn't communicate with my sensor because it had an outdated firmware (V5.2.2).  I updated to firmware V5.2.4 and the problem was solved.  

I would highly recommend mentioning this firmware dependency in the README of this repository.  I would also recommend mentioning that by default the destination device must be configured to have IP address 192.168.11.17 to communicate with the sensor, unless the sensor's default settings are changed.  Both these items caused me much headache and back-and-forth with support.

smartSRA, 2022-09-20:
Great! yes, I've observed a lot of people are having the same issue actually I was not aware that our support has been giving out the old sensor firmware. 

I'll update this in the Readme so others can verify.

> I had this same problem - I couldn't communicate with my sensor because it had an outdated firmware (V5.2.2). I updated to firmware V5.2.4 and the problem was solved.
>
> I would highly recommend [...]

NilaySener, 2022-09-26:
Hi, I tested the package after updating the firmware and the problem was solved. Thank you for helping!