smartmicro/smartmicro_ros2_radars #16: Can not build the workspace

ijklmnopqr, 2022-09-20:
Hello, 

I clone the git repo to my ros2 workspace and colcon build it but it is not compiled and gives the CMake error umrr11_t132_automotivev1.1.1_user_interface
umrr96_t153_automotivev1.2.1_user_interface not found. 
As their dependency is mentioned in smartmicro/smartmicro_ros2_radars/simulator/simulation/CMakeLists.txt. 
Can you please help me with how can I install the radar drivers in ros2? Also, the radar which I am using has the number UMRR-8F0500, does the given drivers support it?

It will be good to have steps for installation process as per ros2 if possible.

smartSRA, 2022-09-21:
Hey,

UMRR8F is not supported by the driver and regarding the build, prior to the build you should attain the smart access release which would provide with the interface files. Need to run the script.

ijklmnopqr, 2022-09-21:
Hello,

So for UMRR87 are the drivers available?

smartSRA, 2022-09-21:
> Hello,
> 
> So for UMRR8F are the drivers available?

Nope. It is not supported.

closed 2022-09-21 by smartSRA

smartSRA, 2022-09-21:
> Will the ros 1 driver support UMRR8F?

Nope, it also does not support the UMRR8F.