smartmicro/smartmicro_ros2_radars #21: node throws "what():  At least one sensor must be configured." error when adding an eighth sensor to radar.template.yaml

chilisam, 2022-11-02:
With up until 7 sensors specified in `radar.template.yaml `, `radar.launch.py` runs just fine but when adding an eighth it thows the error:

```
...
[smartmicro_radar_node_exe-1] terminate called after throwing an instance of 'std::runtime_error'
[smartmicro_radar_node_exe-1]   what():  At least one sensor must be configured.
```

I've made sure that the indentation is correct. Here you can compare `radar7.template.yaml.txt` that works, i.e. the one with 7 sensors, and `radar8.template.yaml.txt` which is the one that breaks, i.e. the one with 8 sensors.

[radar7.template.yaml.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/9922341/radar7.template.yaml.txt)
[radar8.template.yaml.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/9922355/radar8.template.yaml.txt)

smartSRA, 2022-11-09:
Hey,

Thanks for param providing the files. I did reproduce it and yes I observed the problem. I am working on it and would let you know once done. 

A quick solution would be to increase the max count to ten in the following line and than it should work.

https://github.com/smartmicro/smartmicro_ros2_radars/blob/a08720ba2294bb3271b9f24489e8adc4d14528ef/umrr_ros2_driver/include/umrr_ros2_driver/smartmicro_radar_node.hpp#L48.

chilisam, 2022-11-09:
I tried it and now it works, thank you!. 

Note: I had to set the `kMaxSensorCount` to one number higher than the actual number of sensors I'm setting up. To set up 10 sensors (i.e. from 0 to 9) I had to set the value to `11UL`, otherwise it wouldn't work.

smartSRA, 2022-11-09:
Great!. I am going to close this issue for now and update the count in a following patch release. Thanks for reporting.

closed 2022-11-09 by smartSRA

paaraujo, 2023-06-15:
Hi @chilisam, how did you configure the `hw_inventory.json` file to have multiple radars working? Besides the `radar.template.yaml`, what else did you change?