smartmicro/smartmicro_ros2_radars #29: Parsing Error

umsarmy, 2023-02-16:
[smartmicro_radar_node_exe-1] terminate called after throwing an instance of 'nlohmann::detail::parse_error'
[smartmicro_radar_node_exe-1]   what():  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal

smartSRA, 2023-02-16:
Hey,

It seems you are mixing up docker and standard commands which is leading to problem in your file paths. As I am not entirely sure which steps you took to build the driver, delete your “build” and “install” folder and follow the following steps:

1) source your terminal
2) `colcon build` from root of your workspace
3) source your workspace
2) `ros2 launch umrr_ros2_driver radar.launch.py` also called from root of your workspace.

umsarmy, 2023-02-17:
Hi @smartSRA, Thanks for a quick reply.

I followed the streps and the issue got resolved. Thanks!

umsarmy, 2023-02-17:
However, I am not able to see any data on the rostopic

/smart_radar/targets_0
/smart_radar/targets_1
/smart_radar/targets_2
/smart_radar/targets_3

I have uploaded a screenshot of my radar.template.yaml.
I am using UMRR11 Type 132.

[image]

smartSRA, 2023-02-17:
Hey,

The params look fine. Could you verify if you are using a compatible sensor firmware with the driver? The compatible versions are mentioned in the Readme.

umsarmy, 2023-02-17:
Thanks, @smartSRA for the reply.

Actually, I got it worked by just changing the ipv4 address of my laptop from _**192.168.11.99**_ to _**192.168.11.17**_.

Other things remained the same.

umsarmy, 2023-02-17:
@smartSRA Do you why it only works when my system IP is 192.168.11.17? Why I can't use other IP addresses?

smartSRA, 2023-02-17:
> Thanks, @smartSRA for the reply.
> 
> Actually, I got it worked by just changing the ipv4 address of my laptop from _**192.168.11.99**_ to _**192.168.11.17**_.
> 
> Other things remained the same.

Yes, it should have been `192.168.11.17`.

smartSRA, 2023-02-17:
> @smartSRA Do you why it only works when my system IP is 192.168.11.17? Why I can't use other IP addresses?

This is the default ip address for the source in the sensor configuration. The sensor looks for this source when trying to communicate. In order to use another ip address it could be modified. This could be achieved using the driverecorder tool and how to achieve this should be mentioned in the sensor configuration manual.

umsarmy, 2023-02-17:
OK. Thanks

closed 2023-02-17 by umsarmy