smartmicro/smartmicro_ros2_radars #30: PointCloud2 visualization issue

pawelirh, 2023-04-28:
I wanted to run your ros2 driver interfaced with simulated data and visualize it in rviz.
After following instructions from readme and using a predefined Dockerfile and compose file I managed to ros2 topic echo of simulated PointCloud2 data. However, each time when I try to visualize the data in rviz (using radar_rviz.launch.py or standalone rviz in a separate custom container) I encounter the following error:
``
[rviz2-2] rviz2: /tmp/binarydeb/ros-foxy-rviz-ogre-vendor-8.2.8/.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/include/OgreAxisAlignedBox.h:225: void Ogre::AxisAlignedBox::setExtents(const Vector3&, const Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
``

I also proceed with additional tests, I played rosbag with PointCloud from another source and the same rviz instance was able to clearly visualize it.

Did you ever encounter a similar issue?

smartSRA, 2023-05-02:
I have not encountered this issue but I assume the reason for this is that the `simulated data` does not coincide with the pointcloud format of the driver and hence rviz is failing.

Is the `simulated data` also from a sms radar and which type of radar?

pawelirh, 2023-05-04:
By the `simulated data`, I mean data generated by `simulator.cpp`.

I tried to simulate all available sensor configurations accordingly to those listed in compose file (UMRR11, UMRR96, UMRR9D, UMRR9F). However, I paid the most attention to UMRR9F, since I want to simulate the point cloud of DRVEGRD 169 radar. 

Could you maybe provide me with a rosbag file of a point cloud generated by DRVEGRD 169 radar, if a solution of the above problem is more complex?

smartSRA, 2023-05-04:
Yes, in that case it makes sense as to why rviz is complaining. The simulated data is only for testing the functionality of the driver and is not an actual pointcloud relevant data. 

Yes, I believe we have certain recordings for the DRVEGRD 169, I will look into it and see how to make them available to you.

pawelirh, 2023-05-05:
Ok, that makes everything clear. Anyway, thanks for your help. 

Having that rosbag would be enormously helpful for me, let me know if you need any contact details from me.

smartSRA, 2023-05-08:
Hey,

Could you provide me with with your e-mail id so that I can ask my support team to forward the ROSBAGS to you?

closed 2023-05-15 by smartSRA

adithyab94, 2023-10-09:
Hi I use the same sensor and would like to have the rosbag file for understanding the pointcloud data. It would be very helpful if you can send me the rosbag file too 

adithya.balaji@huber-group.com