smartmicro/smartmicro_ros2_radars #32: Driver Configuration

paaraujo, 2023-06-13:
Hi, I could not find in the documentation how I should change the JSON files to make the system work.

I have 4 UMRR-96, I already changed the IP addresses and the sync mode using the DriverRecorder. They are working and synced.

Now, I am trying to collect data using ROS2. My radar.template.yaml file is attached below. By intuition, I thought I had to change some JSON files inside the smartmicro folder as well (also attached).

When I try to launch the code, I get the following error:
`[INFO] [launch]: All log files can be found below /home/paulo/.ros/log/2023-06-13-16-14-16-862773-paulo-GF65-Thin-10UE-36085
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [smartmicro_radar_node_exe-1]: process started with pid [36086]
[smartmicro_radar_node_exe-1] terminate called after throwing an instance of 'std::runtime_error'
[smartmicro_radar_node_exe-1]   what():  Parameter dev_id not set.
[ERROR] [smartmicro_radar_node_exe-1]: process has died [pid 36086, exit code -6, cmd '/home/paulo/workspaces/radars/install/umrr_ros2_driver/lib/umrr_ros2_driver/smartmicro_radar_node_exe --ros-args -r __node:=smart_radar --params-file /home/paulo/workspaces/radars/install/umrr_ros2_driver/share/umrr_ros2_driver/param/radar.template.yaml'].
`

Any help with the setup will be appreciated. I updated the firmware following the instructions provided in the README file. Also, I am using Ubuntu 22.04 and ROS2 Humble Hawksbill. Thank you.

[hw_inventory .txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/11738978/hw_inventory.txt)
[routing_table.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/11738985/routing_table.txt)
[smart_access_config.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/11738992/smart_access_config.txt)
[radar.template.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/11739000/radar.template.txt)

smartSRA, 2023-06-16:
Hey,

There is no documentation available regarding json files as they should not be tempered with. They **only** file that you need to adopt is the `radar.template.yaml` file.

I have seen your attached file and it is not properly configured. Kindly note the following points:

- `master_client_id` should be a unique number and also not 0!
- `hw_dev_id` should be any other number than 0.
- sensor_1 and sensor_0 have the same ip address which would not work. Each sensor should have a unique ip address.
- senor_0 has a id 0 which is not correct. id should be a unique number and not zero.
- Lastly, make sure all ids are unique and not null.

paaraujo, 2023-06-16:
I got some assistance from your reseller, I was missing a few things. First, the `hw_iface_name`, they explained how to get that using `ifconfig`. I guess this is an important thing to be included in the README file. Another thing I detailed in the original question, I was using Ubuntu 22.04 and ROS2 Humble Hawksbill, which seems incompatible. So, I prepared a machine with Ubuntu 20.04 and ROS2 Foxy and after proper configuration, everything worked.

closed 2023-06-16 by paaraujo