smartmicro/smartmicro_ros2_radars #33: Provide support for new sensor types

smartSRA, 2023-06-19:
The new smartmicro sensor "a4" will shortly be available to the customers. This new sensor type should be supported by the driver. Also the DRVEGRD 152 has a new firmware which in return causes new interfaces and hence the driver should be adopted to allow the user support for the new user-interface.

purple-mesh, 2023-06-19:
Also wondering if there is any plans to support Type 42 and Type 48 radars?

smartSRA, 2023-06-20:
> Also wondering if there is any plans to support UMRR-0C Type 42 and UMRR-12 Type 48 radars? Or any suggestions on how one might go about integrating it?

Hey,

The sensor types you mentioned are for the traffic. This repository only supports automotive sensors.
In theory you could follow this repository and instead of automotive sensors use the traffic sensors but for that you would require a smart access for the traffic.

purple-mesh, 2023-06-20:
Thanks @smartSRA! 

I do have the smart access library and can see how it gets used to initialise and parse the data.

Will probably give it a go at integrating it and see how that works out.

smartSRA, 2023-06-21:
Sounds great, do let me know how it goes.

purple-mesh, 2023-07-04:
Hey @smartSRA

Had some time today to tinker with this, happy to report that I have managed to get it working on my end without significant code changes.
Having everything already setup and organised in this repo made this really simple and straightforward.

The brief process I went through was:

- Extract the **traffic version** of the smart access library, in this cased I was using `SmartAccessTm_5_1_0` 
- Update the ros2 node `umrr_ros2_driver/src/smartmicro_radar_node.cpp` file to support this additional radar type (in my case it is the umrr12), don't forget the header file too
- Update the `umrr_ros2_driver/CMakeLists.txt` file so that the corresponding files and libraries are sourced
- Tweak the config files as per `Readme.md`
- For testing I temporarily commented out all the existing supported radar types and just ran with the UMRR12 Type48 radar in simulation mode (enabled through Traffic Management Configurator).

Some minor differences in naming convention:

- The automotive radars have headers such as `Target.h`
- The traffic radars have headers such as `TmTarget.h`
- This `Tm` prefix also extends across various method names, just need to watch out for those.

smartSRA, 2023-07-04:
Hey @purple-mesh ,

Great!! That is exactly the way to go about it and glad to see you could figure out pretty quickly. Regarding the `alive` protocol yes, it is for the automotive sensors not supported and hence was set false which is not the case in traffic sensors.

closed 2023-08-21 by smartSRA