smartmicro/smartmicro_ros2_radars #35: Driver Performance

paaraujo, 2023-07-20:
Hello, I could successfully use the driver to sync four UMRR 96 following all the recommendations available in the README file. After recording some data, I noticed the maximum data rate was only 8.2 Hz. Theoretically, it was supposed to be ~ 16 Hz as the documentation says the triggering interval changes from 55 ms to 60 ms. Is this a limitation of the driver or there is something else to be configured in order to improve this aspect? Or, are the sensors designed like this?

smartSRA, 2023-07-20:
Hey,

Yes, it should ~ 16 Hz. I believe in your case you may have both CAN and ethernet hence the triggering interval doubles and is 120 ms which explains the low rate. You should deactivate CAN, either using Driverecorde tool or follow my PR which implements a new parameter with which we can disable CAN.

paaraujo, 2023-07-20:
Hi @smartSRA , I checked all the sensors again, and I had forgotten to disable it in one of the radars. Now, I am getting ~16.6 Hz! Thank you for the reminder!

Franklein20, 2023-07-20:
Hello @paaraujo 
I'm glad to read that you were able to use the driver.
I've been trying to use this driver with UMRR96  for some time now, but without success. 
I use ros2 humble.

After having cloned and built the package, did you make any modifications to the code or what configuration to perform on the PC?.

Thanks

paaraujo, 2023-07-20:
Hello @Franklein20 , ROS2 Foxy was the only version that worked for me. After moving to Ubuntu 20.04 and ROS2 Foxy, I followed the instructions of the driver, and it worked. I am afraid you won't be able to use Humble.

Franklein20, 2023-07-20:
Ok i will test it.
For the cliend_id, have you put the serial number?

paaraujo, 2023-07-20:
Client_id I just followed the template. I have 4 radars, and I am using 100, 200, 300 and 400.
The tricky part was the `iface_name`, because it wasn't documented. You can get it using `ifconfig`. I mentioned this [here](https://github.com/smartmicro/smartmicro_ros2_radars/issues/32#issuecomment-1594979127).

Franklein20, 2023-07-20:
---
/**:
  ros__parameters:
    # The client_id to be set in smart_access_config.json, must be a unique integer.
      master_client_id: 100
      # This sets the dev_id in the hw_inventory.json
      hw_dev_id: 2
      # This sets the port in the hw_inventory.json
      hw_port: 55555
      # This sets the iface_name in the hw_inventory.json
      hw_iface_name: "eth0"
      # An array of sensors to subscribe to.
      sensors:
          # As many as 10 sensors all named as "sensor_<number>" in increasing order of numbers,
          # e.g., sensor_0, sensor_1, etc. The list must start with sensor_0.
          sensor_0:
              # The model of the connected sensor. Available models umrr11, umrr9d, umrr96,
              # umrr9f_v1_1_1, umrr9f_v2_0_0
              model: "umrr96"
              # The client_id of the sensor/source, must be a unique integer.
              id: 0
              # The ip address of the sensor or of the source acting as a sensor.
              ip: "192.168.11.11"
              # The port to be used.
              iface_name: "eth0"
              port: 55555
              # The frame_id to be set to the published messages.
              frame_id: "umrr"
              # Specify the history size.
              history_size: 10
          `

Franklein20, 2023-07-20:
That is my current config to radar.template.yaml and also the capture of 'ifconfig'
[image]
please could you explain what correspond to the `iface_name`

Franklein20, 2023-07-20:
My current error
[image]

paaraujo, 2023-07-20:
> 

Your iface_name is `enp0s31f6`.

paaraujo, 2023-07-20:
> My current error [image]

This was my error before moving to ROS2 Foxy, and changing the template file as we talked. Solve these points and it will work.

smartSRA, 2023-07-21:
> Hello @Franklein20 , ROS2 Foxy was the only version that worked for me. After moving to Ubuntu 20.04 and ROS2 Foxy, I followed the instructions of the driver, and it worked. I am afraid you won't be able to use Humble.

Hey @paaraujo , it should be compatible with humble and I have tried it myself. Can you tell me what issues you faced while running on humble?

smartSRA, 2023-07-21:
> > My current error [image]
> 
> This was my error before moving to ROS2 Foxy, and changing the template file as we talked. Solve these points and it will work.

The error is not related to foxy or humble, it states that you have not properly configured your sensors. As you can see it says `atleast one sensor must be configured` so your parameters were incomplete or faulty. Also you were able to build on humble, if it was not compatible it would have failed.

Franklein20, 2023-07-21:
Hi @smartSRA 
On the sensor, i set the IP adresse "198.162.11.11"  and also set the IP adress destination "198.162.11.17" 
I put the static IP adress on my PC "198.162.11.17" 
Are there any specific configurations I should make?

smartSRA, 2023-07-21:
Hey @Franklein20,

The default settings for sensor is `192.168.11.11` and has the source address of `192.168.11.17`, so you do not require to set these if you are using the default settings!

Yes, that is right if you are using default settings you only need to set the interface ip address that of the source in default case `192.168.11.17`. To be sure if sensor and source are communicating and sending packets, you can view the traffic using wireshark or make a tcpdump. It is important you make sure that you have established proper connection before running the driver.

Franklein20, 2023-07-21:
Ok thanks.
I modified the configuration of the sensor ( i leave the native configuration) 
[image]
the ouput on windows, i don't what means the info on the left about the cycle time , but the radar seems work

Franklein20, 2023-07-21:
On ubuntu, i checked the connection with the `ping 192.168.11.11` 
That is the output 
[image]
I think the connection is etablished

Franklein20, 2023-07-21:
Despite all this, I still get the same error on the terminal when I run the ros2 driver. 
`---
/**:
  ros__parameters:
    # The client_id to be set in smart_access_config.json, must be a unique integer.
      master_client_id: 100
      # This sets the dev_id in the hw_inventory.json
      hw_dev_id: 2
      # This sets the port in the hw_inventory.json
      hw_port: 55555
      # This sets the iface_name in the hw_inventory.json
      hw_iface_name: "eth0"
      # An array of sensors to subscribe to.
      sensors:
          # As many as 10 sensors all named as "sensor_<number>" in increasing order of numbers,
          # e.g., sensor_0, sensor_1, etc. The list must start with sensor_0.
          sensor_0:
              # The model of the connected sensor. Available models umrr11, umrr9d, umrr96,
              # umrr9f_v1_1_1, umrr9f_v2_0_0
              model: "umrr96"
              # The client_id of the sensor/source, must be a unique integer.
              id: 0
              # The ip address of the sensor or of the source acting as a sensor.
              ip: "192.168.11.11"
              # The port to be used.
              iface_name: enp0s31f6
              port: 55555
              # The frame_id to be set to the published messages.
              frame_id: "umrr"
              # Specify the history size.
              history_size: 10
          
        `

Franklein20, 2023-07-21:
@smartSRA 
Could you help me find where I'm making a mistake?

smartSRA, 2023-07-21:
I would ask you to revise your parameters, I see quite a few mismatches and discrepancies. Also, it would be easier if you could attach while it is bit hard to read in a comment.

Few things, I have noticed; your `hw_iface_name` is incosistent. You only require a `hw_iface_name`, why do you have an extra `iface_name` in your sensor? 
Your `sensor_id` is 0, this is also wrong. Pick any other number than 0 and unique.

You should follow the `radar.template.yaml` file in the umrr_ros2_driver package. Do not add new fields or change names, only set the parameter values according to your settings.

Franklein20, 2023-07-21:
it works on Humble, Thanks 
[image]

But the topic /smart_radar/targets_0 is empty, I don't have a cloud point on rviz

smartSRA, 2023-07-21:
Great!
I need your syslogs to view what is the issue. I would ask you to verify your sensor user_interface and firmware versions. They are mentioned in the Readme section. Only the mentioned versions are compatible with the driver.

So for UMRR96, you should have firmware :V5.2.4 and the user_interface V1.2.1. Please verify this.

Franklein20, 2023-07-21:
Yes the version is ok i use UMRR96 Type 153 
When i launch in the terminal , everything seem good
[image]
the topic is publised but no data

smartSRA, 2023-07-21:
> Yes the version is ok i use UMRR96 Type 153 When i launch in the terminal , everything seem good [image] the topic is publised but no data

Can you attach your `radar.template.yaml` file and your `syslogs`. I would look and only than can I give you a feedback.

Franklein20, 2023-07-21:
[radar.templatef.txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/12124856/radar.templatef.txt)
I don't understand what you call `syslogs` is this file in `/var/log/`

smartSRA, 2023-07-21:
You need to install rsylog in your system and than you should be able to get the syslogs.
In the file you attached your master_client_id and sensor_id is the same! Should be **unique**!

Update your ids and give it a try. If you are still not able to receive data, I would require the syslogs.

Franklein20, 2023-07-21:
the ids is updated, bug stil no data pubished.
[radar.template .txt](https://github.com/smartmicro/smartmicro_ros2_radars/files/12130065/radar.template.txt)
i try to do the syslog, the rsylog is already installed

Franklein20, 2023-07-21:
Unfortunately I can't do syslog. 
Is there anything else you'd like to help me with?

Franklein20, 2023-07-24:
Hello @smartSRA 
I also did a test with docker compose following the instructions in the readme and docker compose up runs for a few minutes without creating containers.

smartSRA, 2023-07-24:
Hey @Franklein20 ,

Unfortunately, you would need to research a bit in how to get the syslogs and send them to me. Without them I cannot provide you with answers. I would still ask you to verify your sensor firmware and user-interface version for UMRR96. At the moment this seems the most possible issue.

So for `UMRR96 T153` you should have firmware `V5.2.4` and the user_interface `V1.2.1`.

Franklein20, 2023-07-24:
https://drive.google.com/file/d/1KARgs-3j9xBJDo7udmEkQQIQGxj-GiDg/view?usp=sharing
the link to find the syslog

smartSRA, 2023-07-24:
[image]

I can see from the logs that the sensor does not have a valid interface. There seems to be an issue with versioning or wrong sensor.

You mentioned you have verified that your sensor user-interface version and firmware version are correct? Can you tell me how did you verify them?

Franklein20, 2023-07-24:
I looked it up on the datasheet. 
Can you tell me how to look with DriveRecord3?

smartSRA, 2023-07-24:
There is a command configurator in the driverecorder. There you can send status request to the sensor. You probably have a manual on how to send commands with driverecorder, there it has been mentioned in detail. I would ask you to get in contact with our support team so they can help you with driverecorder and provide you with a valid/compatible firmware bundle.

The issue is that you do not have compatible versions and hence not being able to communicate with the sensor.

Franklein20, 2023-07-24:
Ok thanks for you assistance

closed 2023-07-24 by smartSRA