smartmicro/smartmicro_ros2_radars #37: how to set destination ip

amdsobhy, 2023-09-07:
How can I set destination ip?

purple-mesh, 2023-09-12:
Hi @amdsobhy

You should be able to connect to your radar by specifying the radar `ip` `port` `hw_iface_name` and the remaining parameters in `umrr_ros2_driver/param/radar.template.yaml`

As far as I can tell, there is no need to specify the destination IP directly, but you should ensure that you have the ethernet connection configured correctly so that you can indeed receive radar packets from the radar.

That is, if radar IP is `192.168.11.11` and the default PC IP is `192.168.11.1` then you need to ensure that the OS ethernet IPv4 setting is configured to `192.168.11.1` with subnet `255.255.255.0`.

amdsobhy, 2023-09-12:
Thank you for your reply. I am currently experiencing a problem where the output on the terminat indicate a successful connection, but when I echo the radar topics or visualize the point cloud I get no data. I do not know why is that

purple-mesh, 2023-09-13:
Hi @amdsobhy

Seems like the problem is fairly broad, but I would suggest the following:

* If you want to be certain that radar is actually sending out packets through ethernet connection, you can try to inspect the ethernet network traffic using packet analyser such as `wireshark`
* If you are not sure whether the target list call back is being called, you may find it useful to modify the source file in `umrr_ros2_driver/src/smartmicro_radar_node.cpp` and update the relevant `targetlist_callback` method(s).

Hope this helps.

pablogago11, 2023-09-14:
Hi all,

We are struggling to connect 2 UMRR-96 Type 153 units ensemble via Ethernet. Our setup is as follows:
- **Host PC**: set IP to 192.168.11.17 (as it is the address the radars look for)
- **Network**: both radars are connected via Ethernet using a [Technica 100BaseT1 adapter](https://www.technica-engineering.com/mc-100base-t1-bcm/) 
- **Radars**: one is set to be Master and the other to be Slave
- **Template file**: the radar.template.yaml file is configured to have 2 radar devices but we achieve the same results as @amdsobhy, terminal outputs that both devices are started, only 1 is publishing (we have slightly modified the node) but the RVIZ does not show any point.

How shall we proceed? Thanks in advance.

purple-mesh, 2023-09-15:
Hi @pablogago11 

Sorry I do not have access to UMRR-96 Type 153 radars, but I would suggest trying to isolate whether the issue resides at the smartmicro driver level or the ros2 driver level.

If you can confirm that the example program `umrr_ros2_driver/smartmicro/example/src/example_app.cpp` in the smartmicro driver package (the one you get from `smart_extract.sh` after it finishes download and extraction) then you can be reasonably sure that they are working fine there and focus on the ros2 component.

For further reference see `umrr_ros2_driver/smartmicro/doc`

smartSRA, 2023-09-22:
@amdsobhy 
Can you comment on your status? were you able to resolve? 

Generally, to set a custom ip address you may use the drive recorder or send a command via the services provided. For successful communication you need to make sure the sensor and the destination are on the same subnet!

Also, make sure that the user interface/firmware of the sensor you are using is compatible with the driver. The list could be found in the 'Readme' section.

pablogago11, 2023-09-25:
Hi all,

I can confirm that with the new release, both umrr96 radars can stream data using a unique ethernet switch. Our procedure was to:
1) Connect only 1 radar (default IP: 192.168.11.11)
2) Launch the ROS2 and call the service as explained in the README to change the IP, i.e. 192.168.11.12
3) Connect both radars to an ethernet switch
4) Emit the tf2_ros static transform from map to umrr frame to visualize the data in the RVIZ

Thanks @smartSRA for the update.

smartSRA, 2023-09-26:
I believe all points have been addressed and I would close this issue for now.

closed 2023-09-26 by smartSRA

Farah-Rostom, 2025-02-17:
I need to change the destination ip i tried to give my laptop an ip of 192.168.11.20 and ist stoped at the on_activate() is called, whe n i changed it again to 192.168.11.17 it worked fine so how can i change the desitantion ip?