smartmicro/smartmicro_ros2_radars #40: Setup UMRR9F Type 169: V2.0.2 in the new release

adithyab94, 2023-10-09:
Hi, 

I use a UMRR9F Type 169: V2.0.2 sensor and I cant see it in the list of avalilable models. How do I setup my sensor. Can I use v2.1.1

https://github.com/smartmicro/smartmicro_ros2_radars/blob/587467f78bc2e64422d4989fc89c025409af8f01/umrr_ros2_driver/param/radar.sensor.template.yaml#L10

smartSRA, 2023-10-10:
Hey,

I would suggest you to get the new firmware v2.2.0 from our support team and than use either `umrr9f_v2_1_1` or `umrr9f_v2_2_1`.

adithyab94, 2023-10-10:
Thank you. Also, We are trying to setup the DRVGRD 169 and meanwhile can you email us a sample rosbag file to get an idea about pointcloud data if possible - [email](mailto:adithya.balaji@huber-group.com)

smartSRA, 2023-10-10:
I suppose that should be available but not sure if through email. Kindly request our support team and ask them for the firmware and rosbags, probably shared through a shared drive.

closed 2023-10-10 by adithyab94