smartmicro/smartmicro_ros2_radars #44: Feature Request: Option to publish Polar Coordinates

adithyab94, 2023-11-03:
Hi,

I have come across a scenario that necessitates the utilization of polar coordinate data (range, azimuth, elevation angles, and noise levels) for specific filtering purposes. Currently, the radar data is published in Cartesian coordinates like in this [example](https://github.com/smartmicro/smartmicro_ros2_radars/blob/master/umrr_ros2_driver/src/smartmicro_radar_node.cpp#L1129). 

For our applications, converting the Cartesian data back to polar form is an extra step that introduces redundancy in the processing chain. It would be significantly more efficient if the driver could provide an option to publish directly the polar coordinates also

Having the ability to access polar coordinate data directly from the driver would streamline our processing workflow and potentially benefit others in the community who might be facing similar requirements.

Thank you for considering this feature request

smartSRA, 2023-11-08:
Hey, 

Feature request noted and the required target values would be available in this next release. I have included it in the PR #43.

smartSRA, 2023-12-20:
The polar co-oridnates are now available and this issue can now be closed.

closed 2023-12-20 by smartSRA