smartmicro/smartmicro_ros2_radars #47: Parsing Error

SalaheldinM, 2024-05-22:
The connection is being done with a CAN with the latest version of the code. We are using foxy and Ubuntu 20.04, we only have one sensor so our yaml file looks something like this:
`/smart_radar:
  ros__parameters:
    adapters:
      adapter_0:
        hw_dev_id: 3
        hw_iface_name: can0
        hw_type: can
        port: 55555
    master_data_serial_type: can_based
    master_inst_serial_type: can_based
    sensors:
      # As many as 10 sensors all named as "sensor_<number>" in increasing order of numbers,
      # e.g., sensor_0, sensor_1, etc. The list must start with sensor_0.
      # For sensor `model`if using :
      # can: 'umrra4_can_v1_0_1', 'umrr96_can', 'umrr11_can', 'umrr9d_can_v1_0_3', 'umrr9d_can_v1_2_2', 'umrr9f_can_v2_1_1', 'umrr9f_can_v2_2_1'
      # port: 'umrra4_v1_0_1', 'umrr96', 'umrr11', 'umrr9d_v1_0_3', 'umrr9d_v1_2_2', 'umrr9f_v1_1_1', 'umrr9f_v2_1_1', 'umrr9f_v2_2_1'
      sensor_0:
        link_type: "can"
        # The model of the connected sensor.
        model: "umrr96"
        # Adapter id to which sensor is connected
        dev_id: 3
        # The client_id of the sensor/source, must be a unique integer.
        id: 200
        # The ip address of the sensor or of the source acting as a sensor.
        ip: ""
        # The port to be used.
        port: 55555
        # The frame_id to be set to the published messages.
        frame_id: "umrr"
        # Specify the history size.
        history_size: 10
        inst_type: "can_based"
        data_type: "can_based"
        # The interface name of the sensor
        uifname: "umrr96_t153_automotive"
        # The major version of the interface
        uifmajorv: 1
        # The minor version of the interface
        uifminorv: 2
        # The pathc version of the interface
        uifpatchv: 2`
        
        but we are running your standard github code, when connected to a can interface we set up using:
        slcand -o -s6 -t hw -S 3000000 /dev/ttyUSBx
        
        and called it can0 and we set this up using this:
        ip link set up my_interface_name
        
        and when we checked the interface `ip link show can0` we saw that it returns this:
```
17: can0: <NOARP,UP,LOWER_UP> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can 

```
we assume from this that we created a can interface properly, when we run the yaml file though it gives this error:

` ros2 launch umrr_ros2_driver radar.launch.py
[INFO] [launch]: All log files can be found below /home/salaheldinm/.ros/log/2024-05-22-10-43-04-363174-salaheldinm-IdeaPad-5-15ALC05-16602
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [smartmicro_radar_node_exe-1]: process started with pid [16604]
[smartmicro_radar_node_exe-1] [INFO] [1716367384.455732336] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.455845470] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.455857513] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.456008479] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.456130369] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.456259714] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1716367384.456389479] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] terminate called after throwing an instance of 'nlohmann::detail::parse_error'
[smartmicro_radar_node_exe-1]   what():  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
[ERROR] [smartmicro_radar_node_exe-1]: process has died [pid 16604, exit code -6, cmd '/home/salaheldinm/smartmicro_ros2_radars-6.0.0/install/umrr_ros2_driver/lib/umrr_ros2_driver/smartmicro_radar_node_exe --ros-args -r __node:=smart_radar --params-file /home/salaheldinm/smartmicro_ros2_radars-6.0.0/install/umrr_ros2_driver/share/umrr_ros2_driver/param/radar.params.template.yaml'].`

we are not sure what we are doing wrong, we are also building how we are supposed to by sourcing terminal first, then doing the docker colcon build, and then sourcing install and then finally running the code. The standard github yaml file doesnt run neither does ours

smartSRA, 2024-05-22:
Hey, 

I see that the CAN setup is good. I would however verify if the sensor is transmitting data with a candump. 

In your approach I see a possible error. I think you have mixed up the docker colcon build and normal colcon build which is causing issue in correctly parsing the yaml file. 

I would suggest to follow the following steps:

1) delete your previous build and install directories. 
2) from the root of your workspace run the command ``colcon build``
3) source your workspace
4) launch the node

SalaheldinM, 2024-05-22:
Hi!

We tried the steps you mentioned based on a previous ticket regarding a similar issue, but they didn't seem to solve the problem. I can see that after running the docker build, the build, install and log folders are locked to only be visible to the root user. Could that be the source of the issue? Since the share config folder is located within install, I was thinking the executable may not be able to access it.

If the config folder that the software works of is not located within install, but the project root's closest config folder, then I suspect the issue might be related to the CMake not generating the smart_access_config.json file.

we also suspect that the can0 interface that we made was a CAN-COM instead of a hardware interface type, could that potentially be an issue?

smartSRA, 2024-05-22:
Hey,

This is something that has been reported before and the reason for this is clear which is the mixing up of docker and native commands. As you use docker build it installs the build files in a different place as to when you use colcon build. Also for consistency and to ensure all dependencies are correctly handled, it is usually better to continue using Docker for running ROS 2 commands if you used Docker for ``building``. This avoids potential issues with environment differences between your host and the Docker container.

Please clarify are you building on a native machine or are you using docker? 

If native, I have already explained in my previous example how to set it up. In case you are working with docker I would than suggest the following steps:

1) docker build as you already have
2) run the docker container interactively
3) navigate to the mounted directory i.e code
4) source ros2 setup files
5) launch the node using ros2 launch

SalaheldinM, 2024-05-22:
when we just clone the repo and do normal colcon build we get errors like for the includes, but we are not sure why they are happening

smartSRA, 2024-05-23:
> when we just clone the repo and do normal colcon build we get errors like for the includes, but we are not sure why they are happening

The reason for that is because you most probably have not extracted the includes. There is a script you need to run as stated in Readme:

[image]

Once you run the script you extract all the includes needed. You shall not get this error.

smartSRA, 2024-05-24:
Hey,

Your assumptions are right, to be sure the sensor is fine I would ask you to monitor the traffic and see if data is being sent out from the sensor. If data is being sent it means the radar is operational and the possible reason for it not publishing targets could be as you stated that the firmware on the sensor is not compatible.

The sensor firmware normally is given out together with the sensor and this is independent of the GitHub repository. If you do not have it I would suggest to get in contact with our support team so they may provide you with it.

SalaheldinM, 2024-05-28:
Hello, we feel like the bit and baud rates may be messed up and that is the reason we get this error when running ip -details -statistics link show :

16: can0: <NOARP,UP,LOWER_UP> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    0          0        1       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    0          0        0       0       0       0    

could you help us with finding the baud rate and bit rate of the radar umrr96 type 153 sensor, we presume that in the command sudo slcand -o -s6 -c -S 500000 /dev/ttyUSB0 can0, the bitrate is s6 but we are also unsure of that. The CAN to USB is not LAWICEL and does not support the -t hw command, it is also not PEAK CAN but we are sure that the firmware should be compatible with your code since the model and the company are in your list of compliant radars. Do you also think that the adapter we are using is causing this issue

SalaheldinM, 2024-05-29:
Hello,
Indeed the issue was due to the adapter, which we replaced with a peak can adapter making us now able to receive the radar signals, thank you for all your efforts. We can close this issue now but we might need help with other matters in the upcoming days. We appreciate your aid.

closed 2024-05-29 by SalaheldinM