smartmicro/smartmicro_ros2_radars #50: DRVEGRD 171 no data on /smart_radar/targets_0

mbellafaire-pme, 2024-05-30:
Hello,

Thank you for the work on this package. I'm currently trying to interface the DRVEGRD 171 with this ROS driver but am not able to view data on the `/smart_radar/targets_0` topic. The radar is connected via ethernet through a media converter and powered by 12V. The node starts up and all services are available:

```
~/workspaces/radar_ws$ ros2 launch umrr_ros2_driver radar.launch.py 
[INFO] [launch]: All log files can be found below /home/mbellafaire/.ros/log/2024-05-30-08-49-34-597791-mbellafaire-Precision-7770-5475
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [smartmicro_radar_node_exe-1]: process started with pid [5478]
[smartmicro_radar_node_exe-1] [INFO] [1717073374.839288010] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1717073374.840248679] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1717073374.840464648] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1717073374.842341874] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1717073376.882388419] [smart_radar]: Data stream services have been received!
[smartmicro_radar_node_exe-1] [INFO] [1717073376.903827229] [smart_radar]: Radar services are ready.
```
My radar parameters file is:
```yaml
---
/**:
  ros__parameters:
    master_data_serial_type: port_based
    master_inst_serial_type: port_based
    
    adapters:
      adapter_0:
        hw_dev_id: 0
        hw_iface_name: enp0s31f6
        hw_type: "eth"
        port: 55555
    sensors:
        sensor_0:
            link_type: "eth"
            model: "umrra4_v1_3_0"
            dev_id: 0
            id: 100
            ip: "192.168.11.11"
            port: 55555
            frame_id: "umrr"
            history_size: 10
            inst_type: 'port_based'
            data_type: 'port_based'
```
Wireshark shows packets being transmitted from the radar to my machine.

[image]

Below is the output for my network adapter obtained through ifconfig:
```
enp0s31f6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.11.17  netmask 255.255.255.0  broadcast 192.168.11.255
        inet6 fe80::4c75:1312:f388:72a9  prefixlen 64  scopeid 0x20<link>
        ether cc:96:e5:7d:94:11  txqueuelen 1000  (Ethernet)
        RX packets 12456  bytes 2790067 (2.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 549  bytes 218097 (218.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  memory 0x90480000-904a0000  
```
Attempting to to echo the `/smart_Radar/targets_0` topic yields no output. Following #15 I requested the firmware from customer support and flashed the radar with version 1.3.0 using this driver.

A similar issue is present on DriveRecorder. The packets are also visible on wireshark for windows, but cannot be viewed. What could be the possible causes for this issue?

mbellafaire-pme, 2024-05-31:
We were able to resolve the issue with the help of customer support. The following parameters needed to be added to the yaml file for DRVEGRD 171:

```yaml 
            # The interface name of the sensor
            uifname: "umrra4_automotive"
            # The major version of the interface
            uifmajorv: 1
            # The minor version of the interface
            uifminorv: 3
            # The pathc version of the interface
            uifpatchv: 0
```

closed 2024-05-31 by mbellafaire-pme

yashgvd, 2024-06-28:
I have this exact same issue. I downloaded the new firmware v1_5_0. But the ROS2 node does not seem to be supporting this version currently.