smartmicro/smartmicro_ros2_radars #52: Configuration of the sensors : the ros2 SetIp service

Farah-Rostom, 2024-08-19:
Hi all,
I tried running the
 `ros2 service call /smart_radar/set_ip_address umrr_ros2_msgs/srv/SetIp "{value_ip: 3232238400, sensor_id: 100}"`
and each time it results in 
`waiting for service to become available... `
and nothing happens my radar is connected to a T1 MEDIACONVERTER that converts can to ethernet my ip is 192.168.11.17 "the recommended ip" and when i run
`sudo tcpdump -i enp3s0` 
this is the output
`14:48:29.694604 ARP, Request who-has 192.168.11.20 tell 192.168.11.17, length 28
`

is it possible for you to walk me through how to get this command to work and what exaclty should I do 
OS: Ubuntu 20.04
ros: foxy

smartSRA, 2024-08-26:
Hey,

There could be a few potential reasons why you might be getting the ``waiting for service to become available...`` message when trying to call the ROS2 service :

Make sure the _**/smart_node**_ is launched and running correctly. It seems that the service is not running: The most common reason is that the service **_/smart_radar/set_ip_address_** is not actually running.

ROS2 environment not sourced: Make sure you've sourced your ROS2 environment in the terminal where you're running the command.

Package not built or installed: Ensure that the _**umrr_ros2_msgs**_ is built and installed correctly.

To troubleshoot:

Check if the service is listed when you run ros2 service list
Verify that the **_/smart_node_** providing the service is running with ros2 node list
Check for any error messages in the terminal where the smart_node is running

After, you have verified and are able to send the service, you would need to adopt your param file with the new ip address that you have allocated. Rebuild and relaunch the package with the new configs for the changes to take place.

Furthermore, I also see an issue here with the tcpdump that you have provided:

The output shows that 192.168.11.17 is not able to find 192.168.11.20 because clearly the ip address has not been set. 

A tipp I would give is to have the driver running with the default ip address of sensor which is 192.168.11.11. Once you are up and running than you can go ahead and make changes. 

The main problem I see here is that the driver is not being properly launched which is also what you have posted in you other issue.

smartSRA, 2024-09-23:
This issue was closed because it has been inactive for over 30 days.

closed 2024-09-23 by smartSRA