smartmicro/smartmicro_ros2_radars #53: Launching the umrr_ros2_driver node

Farah-Rostom, 2024-08-19:
I tried launching the driver ros node but sadly it didn't work, I will give you a quick recap of the steps I followed and my sensor setup.
Setup:
OS: ubuntu 20.04
ros: foxy
Connection: ethernet
host Ip: 192.168.11.17

First, I changed the contents of the .ymal kindly find attached the updated ymal files,taking into consideration I only have 1 radar
[params_ymal.zip](https://github.com/user-attachments/files/16660391/params_ymal.zip)
Then i rebuild the node with the new configurations and ran 
`ros2 launch umrr_ros2_driver radar.launch.py`
which resulted in 
`[INFO] [launch]: All log files can be found below /home/farah/.ros/log/2024-08-19-15-06-17-339655-farah-514329
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [smartmicro_radar_node_exe-1]: process started with pid [514331]
[smartmicro_radar_node_exe-1] [INFO] [1724069177.505200306] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1724069177.505329692] [smart_radar]: Updating ADAPTER params.
[smartmicro_radar_node_exe-1] [INFO] [1724069177.505346223] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1724069177.505487093] [smart_radar]: Updating SENSOR params.
[smartmicro_radar_node_exe-1] [INFO] [1724069179.529806960] [smart_radar]: Data stream services have been received!
[smartmicro_radar_node_exe-1] /home/farah/Application/smartmicro_radar/smartmicro_ros2_radars/umrr_ros2_driver/install/umrr_ros2_driver/lib/umrr_ros2_driver/smartmicro_radar_node_exe: symbol lookup error: /home/farah/Application/smartmicro_radar/smartmicro_ros2_radars/umrr_ros2_driver/install/umrr_ros2_driver/lib/libsmartmicro_radar_node.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_service_type_support_handleIN14umrr_ros2_msgs3srv7SetModeEEEPK29rosidl_service_type_support_tv
[ERROR] [smartmicro_radar_node_exe-1]: process has died [pid 514331, exit code 127, cmd '/home/farah/Application/smartmicro_radar/smartmicro_ros2_radars/umrr_ros2_driver/install/umrr_ros2_driver/lib/umrr_ros2_driver/smartmicro_radar_node_exe --ros-args -r __node:=smart_radar --params-file /home/farah/Application/smartmicro_radar/smartmicro_ros2_radars/umrr_ros2_driver/install/umrr_ros2_driver/share/umrr_ros2_driver/param/radar.params.template.yaml'].
`

Can you please guide me through running the driver using the ethernet connected radar 

Thanks in advance

smartSRA, 2024-08-26:
Hey,

For most part your config seems to be fine however there is a mismatch between the versioning. You have specified model ``umrr9d_v1_0_3`` but you have set the minor version to 1 and patch to 2. It should have been major:1, minor:0 and patch 3.

Also before launching the node, I would stress on verifying that you have established communication between your pc and the radar. Verify this using wireshark or tcpdump.

smartSRA, 2024-09-23:
This issue was closed because it has been inactive for over 40 days.

closed 2024-09-23 by smartSRA