smartmicro/smartmicro_ros2_radars #55: DRVEGRD 171 MSE (ObjectList) Support

sinanyamaan, 2024-09-27:
Hello,

I am trying to create a setup with one **DRVEGRD 171** and two **DRVEGRD 169**s. I have correctly set up the connections via Ethernet and can receive **target pointclouds** from all three sensors.

I am using the umrr9f_t169_msev1.0.0 user interface for the 169s. I chose "mse" because I want to use the detected **objects** from the radars alongside the **targets** in my system.
```cpp
// umrr_ros2_driver/src/smartmicro_radar_node.cpp
if (m_sensors[sensor_idx].pub_type == "mse") {
    m_publishers_obj[sensor_idx] = create_publisher<sensor_msgs::msg::PointCloud2>(
      "smart_radar/port_objects_" + std::to_string(sensor_idx), sensor.history_size);
    m_publishers_port_obj_header[sensor_idx] =
      create_publisher<umrr_ros2_msgs::msg::PortObjectHeader>(
        "smart_radar/port_objectheader_" + std::to_string(sensor_idx), sensor.history_size);
    m_publishers[sensor_idx] = create_publisher<sensor_msgs::msg::PointCloud2>(
      "smart_radar/port_targets_" + std::to_string(sensor_idx), sensor.history_size);
    m_publishers_port_target_header[sensor_idx] =
      create_publisher<umrr_ros2_msgs::msg::PortTargetHeader>(
        "smart_radar/port_targetheader_" + std::to_string(sensor_idx), sensor.history_size);
  } else if (m_sensors[sensor_idx].pub_type == "target") {
    m_publishers[sensor_idx] = create_publisher<sensor_msgs::msg::PointCloud2>(
      "smart_radar/port_targets_" + std::to_string(sensor_idx), sensor.history_size);
    m_publishers_port_target_header[sensor_idx] =
      create_publisher<umrr_ros2_msgs::msg::PortTargetHeader>(
        "smart_radar/port_targetheader_" + std::to_string(sensor_idx), sensor.history_size);
  } else {
    RCLCPP_INFO(this->get_logger(), "Unkwon publish type!");
  }
```

However, I couldn't find any "mse" interface for the 171. There are three options available for the 171: ummra4_v1.0.0, v1.0.1, and v1.2.1, but they are limited to producing only target output.

- Will there be support for retrieving **ObjectList** from the DRVEGRD 171s soon?
- Or is there a solution for this that I might have missed?
- What does "mse" mean in this context, and why is it required for accessing objects?
- Why can't every radar model produce both targets and objects?

smartSRA, 2024-09-27:
Hey @sinanyamaan ,

Great to hear that you have successfully set up the driver and sensors. Regarding your questions:

- The MSE feature for DRVEGRD 171 is currently under development, which is why it's not yet available. While I can't provide a specific release date, it's in the pipeline and should be included in an upcoming driver release.

- MSE stands for Mobile Speed Enforcement, a proprietary software module for specific tracking. Only sensors equipped with the MSE software are capable of tracking and outputting objects; others only publish target data.

- MSE is a specialized software application designed for specific radar types, and not all of our sensors are compatible with it due to varying use cases and hardware limitations.

I hope to have answered all your questions, if you require further details or documentations about MSE you may get in contact with our customer support.

sinanyamaan, 2024-09-29:
Thank you very much for your clear explanations. I am looking forward to the next release!

closed 2024-09-29 by sinanyamaan