smartmicro/smartmicro_ros2_radars #57: Sparse target list on DRVEGRD 169

krzmancd, 2024-10-14:
When running the ROS2 driver, our DRVEGRD 169 returns quite a sparse and incoherent target list. See gif bellow.

[image]

I was expecting a more dense point cloud, that would make more sense. Currently, points are hoping in and out of existence. It's hard to determine objects with confidence on the PCL. Is the problem the fact that I'm testing this indoors with lots of chairs, tables and walls?

My config file is as follows: 

```
/**:
  ros__parameters:
    master_inst_serial_type: "can"
    master_data_serial_type: "can"
    adapters:
      adapter_0:
        hw_type: "can"
        hw_dev_id: 2
        hw_iface_name: "can0"
        baudrate: 500000
sensors:
      sensor_0:
        link_type: "can"
        pub_type: "target"
        model: "umrr9f_can_v2_2_1"
        id: 500
        dev_id: 2
        uifname: "umrr9f_t169_automotive"
        uifmajorv: 2
        uifminorv: 2
        uifpatchv: 1
        frame_id: "umrr"
        history_size: 10
```

Thank you for your help!

smartSRA, 2024-10-14:
Hey,

Your configs look good. The targets are the raw targets from the sensor in pointcloud format whereas the mse/objects are the tracks that are created based on the targets.

Regarding the sparseness, what we see here is expected. Considering you are in your office and in a stationary scenario, a more dense cloud cannot be given out.

For the in- and out-hopping of the targets, this is due to having targets in the same range bin, which are very equal in their attributes. The sensor is re-calculating its noise floor in every cycle, so targets could either fall under the noise threshold in some cycles or change position with equally strong target reflections with a similar radial range. This is a typical behavior of a radar sensor point cloud which cannot be avoided.
 
Compared to Lidar sensors, which give you a much denser point cloud, radar sensors are mostly used to precisely detect the position and the movement of objects, but are not used to get information about the shape of the object, as the resolution is not high enough. The advantages are that the detection of objects is possible within bad lighting situations or bad weather conditions.

I would suggest you try out the sensor with your actual test case and than let me know.

krzmancd, 2024-10-16:
Aha, thanks for your help!

I'll test it in our use case next week. 
Another question. How can I set the operating mode? 
There are 4 modes listed in DRVEGRD 169 sensors datasheet (Long, Medium, Short, Ultra-short).
Can I set them at runtime? 

I also don't understand the `range_toggle_mode` parameter?
Why is a good idea to automatically toggle between modes?
I'm new to radars :)

Thanks!

smartSRA, 2024-10-17:
You can use the ros2 services provided [Modes of Operation](https://github.com/smartmicro/smartmicro_ros2_radars?tab=readme-ov-file#mode-of-operations-of-the-sensors) or you can use the smart_rviz_plugin to set the operating mode.

Regarding the ```range_toggle_mode``` it was a feature that we came out with initially as a customer request and than we continued to support it. The toggle mode as the name suggests switches between different modes long and short. This comes in handy if suppose you have a situation in which normally you are focused on the short range but every now and than you require an input form the long range.

If you would be interested in detailed documentations and further information about our sensors, I would kindly ask you to get in contact with our support team.

Lastly, as I see there are no issues regarding the driver but general discussions. Would it be fine if can we close the issue?

closed 2024-10-17 by krzmancd