smartmicro/smartmicro_ros2_radars #58: Time Synchronization Services

sinanyamaan, 2025-01-20:
I am trying to synchronize total of 6 DRVEGRD line radars within the same ethernet network.

I am following the official documents like Radar_Synchronization_V03, user_manual, etc. They are mentioning that one should add another config within the `smart_acces_config.json` to enable time synchronization. And also create services for master and slaves and register callback for them.

```json
// smart_access_config.json
// ...
"time_sync_cfg": {
    "link_type": "eth",
    "dev_id": 1,
    "role": "master",
    "max_num_of_iter": 50
}
```

As there are no mention of these in the driver's source code, I wonder how would I implement it.
- Are there any examples on time synchronization that I can look into?

I have some ideas on how to implement it in the driver, I wonder if that would work, that I've understood it correctly by reading the documents:
```hpp
// smartmicro_radar_node.hpp
class SmartmicroRadarNode
{
    // ...
    std::shared_ptr<com::master::TimeSyncMasterIface> time_sync_service_master_;
    std::array<std::shared_ptr<com::master::TimeSyncSlaveIface>, detail::kMaxSensorCount - 1> time_sync_service_slave_;
};
```
```cpp
// smartmicro_radar_node.cpp
time_sync_service_master_ = m_services_->GetTimeSyncServiceIface()->GetMasterService();

for ( uint8_t i = 1; i < m_number_of_sensors_; ++i ) {
    time_sync_service_slave_[i] = m_services_->GetTimeSyncServiceIface()->GetSlaveService();
    time_sync_service_slave_[i]->SetSlaveId( i );

    time_sync_service_slave_[i]->RegisterGetSyncTimeCallback( std::bind( ... ) );
    time_sync_service_slave_[i]->RegisterSetSyncOffsetCallback( std::bind( ... ) );

    time_sync_service_slave_[i]->Enable();

    time_sync_service_master_->AddSlave( i );
  }

  time_sync_service_master_->RegisterRefTimeCallback( std::bind( ... ) );
  time_sync_service_master_->Start();
```

- Would an implementation like this work?
- Or do I have to run different `smartmicro_radar_node` for master and slaves?

Your assistance will be very valuable for me to work this out.

smartSRA, 2025-01-21:
Hi,

You’re correct—this feature is not currently available in the driver. However, your approach is on the right track.

You can either modify the existing node to include support for this protocol or create a dedicated node specifically for it.

To assist you further, I recommend referring to the documentation provided with the script `smart_extract.sh` download. In the `smartmicro` folder, navigate to the `doc` section and review the `user_manual.pdf`. The setup for the **time sync protocol** is explained under the **Communication Services** section.

Additionally, for a practical example, you can check the `example_app.cpp` file located in the **src** folder within the example directory of the same `smartmicro` folder. This file demonstrates how to implement the protocol.

Let me know if you need further assistance!

sinanyamaan, 2025-01-24:
Thank you for your detailed explanation.

I followed the documents you mentioned for Time Synchronization. I divided the six radars into two groups, 4 and 2, and configured their Start Radar Synchronization and Time Synchronization among each other.

While I was comparing the incoming packets on Wireshark and the timestamps of PointCloud2 msgs generated by the driver, I saw that the driver uses the timestamp from PortHeader to stamp point clouds. As far as I understand, this value is based on an internal clock that starts counting from the moment the radar is powered on. As a result, the timestamps of the point clouds were always starting from 0.
In my own code, I calculate this data using the timestamp from TargetListHeader. This adjustment ensured that the radars began producing synchronized point clouds.

Also, thank you for reminding me about the example; it had completely missed it.
However, even after building the example according to my computer architecture, I keep encountering a segmentation fault at the same point. Specifically:
```cpp
services->GetTimeSyncServiceIface()->GetMasterService();
```

To make sure, I immediately tested the following:
```cpp
auto time_sync = services->GetTimeSyncServiceIface(); // -> works fine
auto master_service = time_sync->GetMasterService(); // -> segmentation fault occurs here
```

Since the error specifically occurs when calling GetMasterService(), I’m wondering if there’s a missing .so file in my setup.

smartSRA, 2025-01-30:
Hey,

If you are using the example app setup, you will find a CMakefile there in which the libs are linked to the example_app. You should make sure that you also have linked these libs. Unfortunately, at this moment I cannot deduct why this would be happening as I would require more information. To diagnose further I need to know how your configs look like, the syslogs (the smart access writes logs to the syslog) and also your implementation.

On your side my suggestions to tackle this would be first use the example_app and see if you are able to start up the service. Also, I would suggest to reduce complexity by using one one set of sensor combination. If you encounter the issue with the base setup than we could investigate further and would require more inputs from your side.

sinanyamaan, 2025-02-03:
Okay, after too much trials, I've figured that I must put that `time_sync_cfg` block into my `smart_access_config.json`.
```json
"time_sync_cfg": {
    "dev_id": 0,
    "link_type": "eth",
    "role": "master"
  }
```
With this block, my code successfully called `GetMasterService()`. And I was able to register the `RefTimeCallback`. But I believe it didn't triggered, I've added print loggers in the function and didn't see any one of them.
- While everything is correctly set, what doesn't it call the callback? Did I miss something again?

I believe that worked because of `"role: "master"` line. But what can I do for slaves? I can get the interface, but gives a segmentation fault on the `SetSlaveId()` method.
- I can not add another "time_sync_cfg" block in that JSON file.
- There is no example for the slaves in the `example_app`.
- Could not find any example configs in the official smartmicro documentations other than the smart_access_config.json.

This codebase seems like it is modular between different series of radars but it is really easy to make mistakes, forget something and kind of impossible to figure out what. I wonder does it have to be like that?

Anyway, I am kindly asking your guidance on this topic once again.

smartSRA, 2025-02-04:
Hey,

Great to see that you are making progress. You set the slaves by configuring them with parameters. You have to follow the `Radar_Synchronization_V03, user_manual`. There can only be one master, so if you are setting smart access config to master you need to set all other sensors to slave. In the app you just add the slaves as is shown in the example and not in the configs. This could possible be the problem and by configuring so you should be able to receive your callbacks.

The issue with segmentation fault is more complicated and would need as also mentioned in my previous comment more input from your side i.e logs, your implementation..

Also as this issue is not related to the ros2 driver package we would need to investigate it on a different channel.

Please try with the suggested and if you cannot come forward, we would need investigate on our side.

closed 2025-02-14 by smartSRA

krzmancd, 2025-05-07:
Hey @sinanyamaan 

Did you ever figure this out?
I think adding each time sync slave service to a master isn't the correct way, as GetSlaveService() returns a singleton instance. So every std::shared_ptr in time_sync_service_slave_ is the same object. Am I wrong?
Other than that, I came to the same conclusion, but the sync isn't working.