smartmicro/smartmicro_ros2_radars #60: Driver Configuration and parameters

Farah-Rostom, 2025-02-17:
I'm working with the umrr9f_t169_automotive radar using ethernet interface, but I 'm facing 2 issues 

First:

- I need to **change the destination ip**, i tried changing my latop IP to 192.168.11.20 and the driver seems to be stuck at the `on_activate() is called.` when i went back to the 192.168.11.17 ip it worked fine. I need to change the destination IP to be able to recieve the data on a switch.

Second :

- When I noticed the maximum data rate was only 8.2 Hz, it was mentioned i can change this by **deactivating the can interface** Through the tool or the PR parameter can you please elaborate more on the PR paramater as i can't find it

smartSRA, 2025-02-19:
> I'm working with the umrr9f_t169_automotive radar using ethernet interface, but I 'm facing 3 issues
> 
> First:
> 
> * I need to **change the destination ip**, i tried changing my latop IP to 192.168.11.20 and I changed the value of ip_dest_address int the "auto_interface_0dim", but the driver seems to be stuck at the `on_activate() is called.` when i went back to the 192.168.11.17 ip it worked fine. I need to change the destination IP to be able to recieve the data on a switch.
> 
> Second :
> 
> * When I noticed the maximum data rate was only 8.2 Hz, it was mentioned i can change this by **deactivating the can interface** Through the tool or the PR parameter can you please elaborate more on the PR paramater as i can't find it
> 
> Third:
> 
> * when i try to run the setIp service with these values `ros2 service call /smart_radar/set_ip_address umrr_ros2_msgs/srv/SetIp "{value_ip: 3232238398, sensor_id: 100}"`
>   it returns  `The passed service type is invalid`

**First**;

To be able to receive data on desired destination you require to initially use the default destination of ``192.168.11.17``. After sending a `ip_dest_address` instruction you need to send another instruction which saves the first instruction using `comp_eeprom_ctrl_save_param_sec, 2010`. Than you need to restart the sensor.

I did try to reproduce the issue you are facing and I have identified an issue which could be causing problem with the driver. For the time being till I fix it, I would suggest to use our DriveRecorder tool to accomplish this task.

**Second**;

To deactivate CAN you can use the `output_control_target_list_can` instruction with value of 0.

**Third**;

I was unable to reproduce this and would suggest to remove your old build and have a clean build. Seems to be a ros environment issue and check if you have sourced properly.

Farah-Rostom, 2025-02-20:
i used the toool to change the ip_source_dest to 192.168.11.25 and i restarted the radar, i can ping it from my terminal but I can't access the new ip in the tool
I changed the ethernet UDP connection setting to 192.168.11.25 but it didn't make any difference now i can only ping the radar but I can't access it from the tool

smartSRA, 2025-02-20:
You need to update your server settings in the Ethernet UDP to the destination address and in the settings for the devices in UDP mode you need to update the ip address of the sensor.

Farah-Rostom, 2025-02-20:
I already updated the server settings and i didn't change the destination ip i just changed the source

smartSRA, 2025-02-20:
[image]

This is the destination address so you have not changed it.

For the device, you need to update here:

[image]

Farah-Rostom, 2025-02-20:
I changed the 192.168.11.11 to 192.168.11.25 and nothing happened

Farah-Rostom, 2025-02-20:
and i tried accessing it through can and can is not detected so How is this possible?? are there any steps I should follow to modify the can to bs can up??

smartSRA, 2025-02-20:
> I changed the 192.168.11.11 to 192.168.11.25 and nothing happened

Try to reinit  a few times. If not successful, try restarting. Sometimes, it can have difficulties establishing proper connection.
Also, make sure that your computer ethernet interface has the destination address `192.168.11.17`.

smartSRA, 2025-02-20:
> and i tried accessing it through can and can is not detected so How is this possible?? are there any steps I should follow to modify the can to bs can up??

You should be able to send instructions to the sensor with CAN even when you have disabled it. You need to click `reinit` and you should see the green indicator light up and at the same instance send the instruction to reactivate CAN.

Important is to send asap you click reint and indicator turns green.

Farah-Rostom, 2025-02-20:
The thing is when i do candump on my ubuntu i recieve data but when i open the windows and open the tool no data is sent even when i reinit the can

smartSRA, 2025-02-20:
I would suggest that you contact our support team and have a live session. They would help you out to set it up and would be much quicker. It is tough to debug from here.

Farah-Rostom, 2025-02-20:
I sent an email to eu-support@smartmicro.de but no one replied

smartSRA, 2025-02-21:
I asked our support to contact you and they updated me that the issue has been resolved. I am closing this issue now.

closed 2025-02-21 by smartSRA