smartmicro/smartmicro_ros2_radars #64: Cannot receive data from UMRR96-153 via CAN using ROS 2 driver, despite correct CAN signal

Batbedat, 2025-07-01:
Hello,

I am currently getting started with the "umrr96-153" radar for future integration on a mobile robot. I am connecting the radar to the ROS 2 driver via a CAN bus. It is connected on can0, and CAN communication is functioning properly, as verified with candump.

However, the umrr96-153 sensor does not produce any readable data through the ROS 2 driver, even though the CAN interface is active and transmitting. The node does not publish anything.

Below is the relevant section of our configuration:
/smart_radar:
  ros__parameters:
    adapters:
      adapter_0:
        hw_dev_id: 11
        hw_iface_name: can0
        hw_type: can
        port: 500000
    master_data_serial_type: can_based
    master_inst_serial_type: can_based
    sensors:
      sensor_0:
        link_type: "can"
        pub_type: "target"
        model: "umrr96_can_v1_2_2"
        dev_id: 11
        id: 400
        frame_id: "umrr"
        history_size: 10
        inst_type: "can_based"
        data_type: "can_based"
        uifname: "umrr96_t153_automotive"
        uifmajorv: 1
        uifminorv: 2
        uifpatchv: 2

Additionally, according to the logs for the "radar_node.cpp" file, I am getting an error code 2: "Failed to register CAN targetlist for sensor umrr96_can_v1_2_2", which seems to prevent access to the callback.

Question:
Could this mean that my configuration is not suitable ?

Any suggestions or clarifications would be greatly appreciated.

Thanks in advance for your support!

smartSRA, 2025-07-02:
Please try out with the following configuration and also verify the firmware version on the sensor is compatible [version](https://github.com/smartmicro/smartmicro_ros2_radars#:~:text=153%20AUTOMOTIVE%20v1.2.2-,UMRR96%20Type%20153%3A%20V5.2.4,-UMRR11%20Type%20132).

```
ros__parameters:
  master_data_serial_type: can_based
  master_inst_serial_type: can_based
  adapters:
    adapter_0: 
      hw_dev_id: 11
      hw_iface_name: "can0"
      hw_type: "can"
      baudrate: 500000 # You are missing this parameter
  sensors:
    sensor_0:
      link_type: "can"
      pub_type: "target"
      model: "umrr" 
      dev_id: 11
      id: 400
      frame_id: "radar_sensor_rear_left"
      history_size: 10
      inst_type: "can_based"
      data_type: "can_based"
      uifname: "umrr9f_t169_automotive"  
      uifmajorv: 2
      uifminorv: 2
      uifpatchv: 1
```

Batbedat, 2025-07-02:
Thank you for your reply.
My model is the UMRR96-153 radar running firmware 5.4.1.1, but according to the documentation the latest available firmware is UMRR96 Type 153 V5.2.4. I therefore concluded that I should use the most recent User Interface version (UMRR96 Type 153 AUTOMOTIVE v1.2.2). 

I tested the configuration you suggested, but I’m still encountering the same problem: the ROS 2 node produces no output.

Batbedat, 2025-07-03:
Indeed, I needed to download firmware from a version earlier than my radar’s (5.2.1).
Thank you for your help.